frequenz-floss/frequenz-api-microgrid #6: Improve README to include examples or a better description of the project

leandro-lucarella-frequenz, 2022-08-30:
The README will now be what users see first when visiting the PyPi package page, so we should make sure it is a bit more informative to end users (so far it only has information for contributors, not users).

tiyash-basu-frequenz, 2024-05-13:
Is this issue resolved now?

llucax, 2024-05-14:
I guess before releasing a 1.0 it might be good to go through the docs and probably make some improvements, like adding some example, more like the [channels user guide](https://frequenz-floss.github.io/frequenz-channels-python/v1.0/user-guide/quick-start/) (although if we consider only the API spec part it might make less sense). That said, we can also create more specific issues. I leave it to you to close or leave as you prefer.

tiyash-basu-frequenz, 2024-05-14:
Good point. Right now, I feel it will be easier to manage if the requirements are listed in a single issue (here). So, here is a short list of requirements copy-pasted from what you wrote above (please feel free to extend it):

- [ ] Add usage examples (like the [channel user guide](https://frequenz-floss.github.io/frequenz-channels-python/v1.0/user-guide/quick-start/))