frequenz-floss/frequenz-api-microgrid #41: Add a CHP component type

sahas-subramanian-frequenz, 2023-04-19:
### What's needed?

We need to be able to identify the CHP meter from the component graph.

/cc @matthias-wende-frequenz @thomas-nicolai-frequenz 

### Proposed solution

Add a new CHP component type in the microgrid API,  such that the sdk can identify the CHP meter, as the predecessor of the CHP.

This needs to be done on a fork of v0.11.0, so that we can get it can get support for it quickly in the sdk.

### Use cases

The PV optimization actor that is currently being developed.

### Additional context

/cc @matthias-wende-frequenz @thomas-nicolai-frequenz

tiyash-basu-frequenz, 2023-04-19:
> This needs to be done on a fork of v0.11.0, so that we can get it can get support for it quickly in the sdk.
> The PV optimization actor that is currently being developed.

Will the PV optimization actor be ready to use it soon? If not I'll suggest to add it to the v.012.0 release, which can be released by the end of this week.

sahas-subramanian-frequenz, 2023-04-19:
> Will the PV optimization actor be ready to use it soon?

@matthias-wende-frequenz could you confirm?  If we wait for all the changes, then we need this week to release the API, then a few more days, likely a week for the SDK to migrate to the new API, reviewed and merged.

tiyash-basu-frequenz, 2023-04-19:
BTW, I do not think we need anything here for adding a CHP unit, since it already exists as a component category.
I'll confirm and update this issue.

sahas-subramanian-frequenz, 2023-04-19:
> BTW, I do not think we need anything here for adding a CHP unit, since it already exists as a component category.

oh you're right, thanks.  We just need to add support in the sdk.  I will close this issue then.

closed 2023-04-19 by sahas-subramanian-frequenz