frequenz-floss/frequenz-api-microgrid #76: Use `Location` from the common specs

tiyash-basu-frequenz, 2023-08-01:
### What's needed?

The `Location` message defined [here](https://github.com/frequenz-floss/frequenz-api-microgrid/blob/e48d52fb96536fc6bb9422c4b2ca666bcc1a6761/proto/frequenz/api/microgrid/microgrid.proto#L317-L323) should be removed, and instead the one from the common specs should be  used.

### Proposed solution

The following should be used from the common specs repo:
```proto
// A pair of geographical co-ordinates, representing the location of a place.
message Location {
  // The latitude of the place.
  float latitude = 1;

  // The longitude of the place.
  float longitude = 2;
}
```

thomas-nicolai-frequenz, 2023-08-04:
@tiyash-basu-frequenz I think we discussed that we might want to add the country code here as well, right?

```Patch
// A pair of geographical co-ordinates, representing the location of a place.
message Location { 
  // Latitude ranges from -90 (South) to 90 (North)
  float latitude = 1;

   // Longitude ranges from -180 (West) to 180 (East)
   float longitude = 2;
   
+   // Country ISO 3166-1 Alpha 2
+   string country_code = 3;
}
```

@stefan-brus-frequenz FYI

tiyash-basu-frequenz, 2023-08-04:
> I think we discussed that we might want to add the country code here as well, right?

Yep, this PR was just intended to move the existing definition, but I can add another commit here to add the country code.

tiyash-basu-frequenz, 2023-08-09:
Was addressed in https://github.com/frequenz-floss/frequenz-api-microgrid/pull/99

closed 2023-08-09 by tiyash-basu-frequenz