frequenz-floss/frequenz-api-microgrid #178: Add Filter message and add it to the `ReceiveComponentDataStreamRequest`

matthias-wende-frequenz, 2023-11-14:
### What's needed?

The `ReceiveComponentDataStream` rpc returns a stream of responses each of which containing a list of `metric_samples` in the data. The only parameter that can be passed is the `component_id` which implies that the returned stream contains all metrics that are streamable by this components.

From a user perspective not all of these metrics are used and thus are ignored. Still the user has to iterate over the full list. 

### Proposed solution

Instead the users could provide a filter with only the metrics that they need.

### Use cases

Creating high level objects in the SDK that stream composed data but aren't interested in the full data stream. Among the metrics that might not be of interest are all kinds of harmonics, (re)active energy, power factor, total harmonic distortion.

matthias-wende-frequenz, 2023-11-14:
@tiyash-basu-frequenz I've assigned you to the issue for now.
@thomas-nicolai-frequenz FYI

thomas-nicolai-frequenz, 2023-11-14:
> Creating high level objects in the SDK that stream composed data but aren't interested in the full data stream

Adding such a feature would have quite some consequences for the SDK and any apps using multiple actors that require different metrics. Keep in mind that standard apps should included a list of default actors. As a result the SDK would need to be aware about the metrics needed by every actor and relay this to the microgrid API. Second whats the benefit of this optimisation and why is this required right now?

matthias-wende-frequenz, 2023-11-15:
This has been discussed outside of this issue and we discussed applying the same whitelist filtering as in the reporting API.