frequenz-floss/frequenz-api-microgrid #223: Add RPC to specify an additional payload to stream alonwith component data

tiyash-basu-frequenz, 2024-03-22:
### What's needed?

We need an RPC to specify an additional payload to stream alongwith component data. This payload would be overwritten by the next call to this RPC.

### Proposed solution

One solution would be to add the following RPC with the specified request message:

```proto
service Microgrid {
  ...
  rpc AddAdditionalPayload(AddAdditionalPayloadRequest) returns (google.protobuf.Empty)
}
```
```proto
message AddAdditionalPayloadRequest {
    google.protobuf.Struct payload = 1;
}
```

The service would attach this payload to the outgoing data message according to the specs defined in frequenz-api-common.

### Additional context

https://github.com/frequenz-floss/frequenz-api-common/issues/207

llucax, 2024-03-28:
Maybe it would be worth adding something to the use case section of the issue, at least for me it is not clear what is the intended use of this feature.

tiyash-basu-frequenz, 2024-03-28:
> Maybe it would be worth adding something to the use case section of the issue, at least for me it is not clear what is the intended use of this feature.

Already mentioned:
"We need an RPC to specify an additional payload to stream along with component data."

llucax, 2024-03-28:
For me that describe what the feature does, but not why it is needed. Why do we need specify an additional payload to stream along with component data? Who's going to use it and for what?

tiyash-basu-frequenz, 2024-03-28:
This was mentioned in the linked issue. I copy pasted the text from there.

llucax, 2024-03-28:
> This was mentioned in the linked issue. I copy pasted the text from there.

OK, I read the linked issue too but I still find it difficult to understand or see the use case, but maybe it is me just lacking some additional context.

> (While we are discussing this, I will also mention that the issue template feels odd to me. The Use-Case should be present before the Solution szection. I generally add the "why" in the "what's needed" section, because mentioning it after proposing a solution feels odd.)

I'm not surprised, I created the template a long time ago as something experimental and wasn't improved since then :D.

That said, I think moving the use case before the proposed solution might be a bit noisy, at least in my experience use case is empty in 90% (?) of the cases, so it will just be distracting. I [suggested GitHub to add an option to skip empty sections](https://github.com/orgs/community/discussions/63402#discussioncomment-8477912), which really don't make sense to put in the issue at all, but I doubt it will be implemented soon (or at all), so we could explore with different template approaches to avoid the noise.

tiyash-basu-frequenz, 2024-03-28:
>OK, I read the linked issue too but I still find it difficult to understand or see the use case, but maybe it is me just lacking some additional context.

Okay, fair enough, I will try to extend the issue with more context.

>so we could explore with different template approaches to avoid the noise.

Sounds good.

Marenz, 2024-04-08:
To me that issues is still confusing. Who would be the sender and who the receiver of that data? How often would it be sent?

tiyash-basu-frequenz, 2024-04-08:
> Who would be the sender and who the receiver of that data?

Users, with their SDK apps. This data will also be available via the Reporting API.

> How often would it be sent?

As frequently as they want.

> To me that issues is still confusing.

Think of it as a user-specified tag. E.g.,
* users could update the tag after issuing charge/discharge commands. That would tell them whether the data they receive is a result of their command or not.
* this could also be used to stream arbitrary data through the Microgrid API/Reporting API. This about the working point from the FCR actor. ATM, there is no metric that captures it. Something like this would open up the API to send arbitrary tags, so that users can publish any data they want.

llucax, 2024-04-08:
OK, now that I'm not the only one that doesn't understand this feature, I will keep pushing a little bit further, as for me it is still unclear how this is supposed to work exactly and what are real use ceases. For me it would be good to have some pseudo code but real example, like, I don't know, for FCR:

```py
working_point = calculate_working_point()
client.charge(battery_id, tags={"working_point": working_point})
# now what?
async data for client.get_data(battery_id, "soc"):
     # data here will have somehow the tags attached here?
     # How is FCR supposed to use this information?
```

Maybe we should have a meeting to discuss/explain this feature better before moving forward?

tiyash-basu-frequenz, 2024-04-08:
sure

tiyash-basu-frequenz, 2024-04-08:
I got this feedback from @Marenz in a call:

Why stream the data back with the component data stream? Why not have a separate RPC for it?

(I will get to it later, after giving it some thought)

tiyash-basu-frequenz, 2024-04-08:
Feedback from @llucax:

Why have this feature in the Microgrid API at all? Just for convenience?