frequenz-floss/frequenz-api-microgrid #228: Report unused imports in CI

tiyash-basu-frequenz, 2024-03-26:
### What's needed?

We have seen at least once that our protobuf files had unused imports. This did not cause any issue with Rust or Python toolcahins, but caused issues with a Go toolchain (used for building the gRPC gateway). This needed to be [fixed](https://github.com/frequenz-floss/frequenz-api-microgrid/pull/225).

We need the CI to catch such unused imports, so that
a. our toolchains do not break
b. our proto definitions stay clean

The CI should also fail on finding unused imports.

### Proposed solution

The first solution that comes to mind is to run protoc to generate Go code, since that definitely broke. Howeverm there could be other solutions, including the linter. We need to check.

tiyash-basu-frequenz, 2024-04-08:
The path of least resistance right now is to add it as a separate step in the CI file.

llucax, 2024-04-08:
Just to have it in mind for the future, there is a `protolint` issue for implementing this:

* https://github.com/yoheimuta/protolint/issues/125

The idea is accepted, it is just waiting for someone to do it (marked as a "good first issue").