frequenz-floss/frequenz-api-microgrid #260: Remove `Metadata` from `GetMicrogridMetadata` and `GetMicrogridMetadataResponse`

llucax, 2024-09-16:
### What's needed?

It seems the "metadata" word was removed from all other parts, like the `microgrid` message type, so it looks like including `Metadata` in these names is inconsistent.

### Proposed solution

Rename:

* `GetMicrogridMetadata` -> `GetMicrogrid`
* `GetMicrogridMetadataResponse` -> `GetMicrogridResponse`

tiyash-basu-frequenz, 2024-09-16:
Maybe

`GetMicrogridMetadata` => `GetMicrogridInfo`

llucax, 2024-09-17:
For me this is a general design question, and for me the number 1 priority is to have consistency, for me it would be weird that `GerMicrogridInfo` has a `Microgrid` message in the response instead of a `MicrogridInfo` message, so if we name this `GetMicrogridInfo`, I would then rename `Microgrid` to `MicrogridInfo`. And that goes for all messages (`Component` -> `ComponentInfo`, `Battery` -> `BatteryInfo`, etc.).

At this stage the least breaking change would be to just use `GerMicrogrid`. If we are keeping the *inconsistency* of having a method `GetMicrogridInfo` that returns a `Microgrid`, then I don't see a point in renaming `GerMicrogridMetadata` in the first place.

But maybe I'm too of a purist :)

tiyash-basu-frequenz, 2024-09-17:
IMO we should name `Microgrid` to `MicrogridInfo` everywhere.

llucax, 2024-09-18:
Sounds good to me! What about components?

closed 2025-07-07 by tiyash-basu-frequenz