frequenz-floss/frequenz-api-microgrid #268: Rename `AddComponentBounds` to `AugmentComponentBounds`

tiyash-basu-frequenz, 2024-11-06:
### What's needed?

This RPC, for a given component and metric,
- receives a vector of bounds from a client,
- adds it to the existing set of bounds,
- intersects all the bounds in the set,
- sets the resulting pair of bounds as a the final bounds of the component's given metric,
- and finally removes the received pair of bounds from the set (after a user-provided duration), recomputes the final bounds again, and updates it in the component.

The verb `Add` might not accurately represent this action, and could be misleading.

### Proposed solution

Perhaps `Augment` is a better verb?

Also, it looks like the documentation needs to be improved to be more accurate.

llucax, 2024-11-06:
Input from ChatGPT:

---

"Augment" is better than "Add" but still may not fully capture the idea of both adding and then later removing or recomputing after a duration. Here are a few alternatives to consider:

1. **ApplyBounds**: This verb emphasizes that bounds are not just being added; they’re being *applied* to the metric, intersected, and will later be removed.
   
2. **ImposeBounds**: This choice suggests enforcing bounds temporarily, implying that they may change over time.
   
3. **ExtendBounds**: "Extend" implies enhancement or adjustment, with potential to reconfigure the bounds as new inputs are applied.
   
4. **EnvelopBounds**: "Envelop" could work, suggesting that bounds are temporarily wrapping or containing the metric’s state.
   
5. **OverlayBounds**: If the bounds are temporary and layered, "Overlay" reflects that they are supplementary and will be removed after the specified duration.

Each option has a subtly different implication, but **ApplyBounds** and **OverlayBounds** seem most aligned with the behavior you described.

tiyash-basu-frequenz, 2024-11-06:
I do not see how `Apply` is a better choice than `Augment`. `Augment` suggests that it is extending the bounds, which `Apply` does not.
I also do not see any advantage of `Apply` over `Add`.

Let's maybe sleep over it for a day or two. We can close the issue if needed.

thomas-nicolai-frequenz, 2025-07-07:
If I'd have to choose here would be my favourites with augment being at the top of the list. Maybe `ExtendBounds` could also be an option but its a little longer.

1. Augment
2. Add
3. Apply

tiyash-basu-frequenz, 2025-07-08:
I will soon submit a PR renaming the RPC to `AugmentElectricalComponentBounds`.