frequenz-floss/frequenz-api-microgrid #282: Specify data guarantees for live data streaming

llucax, 2025-02-13:
### What's needed?

The live data streaming RPCs are using the [`frequenz.api.common.v1.microgrid.components.ComponentData`](https://github.com/frequenz-floss/frequenz-api-common/blob/c46ef7b25d7d4e6fef0758f71d24536760307665/proto/frequenz/api/common/v1/microgrid/components/components.proto#L246-L255) message as the response, but this message is very flexible to allow streaming data efficiently in many different use cases (like streaming historical data).

Supporting all the provided flexibility is difficult for clients, so if there are some data restriction guarantees, these should be properly documented in the RPC response documentation so clients don't need to include unnecessary extra complexity.

### Proposed solution

Document the [`ReceiveComponentDataStreamResponse.data`](https://github.com/frequenz-floss/frequenz-api-microgrid/blob/f31edd12da90087a2c5f24130c80495aaf1d0323/proto/frequenz/api/microgrid/v1/microgrid.proto#L412C3-L412C60) field to mention that for a single response for a **live data** message:

* All the present [`data.metric_samples`](https://github.com/frequenz-floss/frequenz-api-common/blob/c46ef7b25d7d4e6fef0758f71d24536760307665/proto/frequenz/api/common/v1/metrics/metric_sample.proto#L187-L240) and [`data.states`](https://github.com/frequenz-floss/frequenz-api-common/blob/c46ef7b25d7d4e6fef0758f71d24536760307665/proto/frequenz/api/common/v1/microgrid/components/components.proto#L460-L492) will have the same 'sampled_at` timestamp.
* There will be no duplicate [`metric`](https://github.com/frequenz-floss/frequenz-api-common/blob/v0.x.x/proto/frequenz/api/common/v1/metrics/metric_sample.proto#L192)s in the `data.metric_samples`.
* There will be at most one [`data.states`](https://github.com/frequenz-floss/frequenz-api-common/blob/c46ef7b25d7d4e6fef0758f71d24536760307665/proto/frequenz/api/common/v1/microgrid/components/components.proto#L254).[^1]

[^1]: @tiyash-basu-frequenz could there be none? Or is it guaranteed to have at least one? In any case I don't think it makes a huge difference for the client implementation if it can be 0 or 1 or if is guaranteed to have 1, so we can leave it open.

The same restriction should documented for [`ReceiveSensorDataStreamResponse.data`](https://github.com/frequenz-floss/frequenz-api-microgrid/blob/f31edd12da90087a2c5f24130c80495aaf1d0323/proto/frequenz/api/microgrid/v1/microgrid.proto#L446), adapting to the particular fields of that response.

### Use cases

Live data streaming client implementation.

shsms, 2025-02-19:
> * There will be at most one [`data.states`](https://github.com/frequenz-floss/frequenz-api-common/blob/c46ef7b25d7d4e6fef0758f71d24536760307665/proto/frequenz/api/common/v1/microgrid/components/components.proto#L254).

But there will also be a relay state for batteries, right? Or would that come in a separate message?

llucax, 2025-02-20:
My understanding is no. All states come in one `ComponentState` message, relay states are also `ComponentStateCode` which comes as a `repeated` field. But @tiyash-basu-frequenz should probably confirm or deny.

shsms, 2025-02-20:
Yes, so there won't be "at most one" state, right?  There will be two `ComponentStateCode` values in the `repeated` field.

llucax, 2025-02-20:
Yeah, `ComponentState` != `ComponentStateCode`, so one `ComponentState` that contains many `ComponentStateCode`s.

But, I just talked to @tiyash-basu-frequenz and apparently it is not true, as relays and batteries can report data separate so have different timestamps, so maybe we'll have 2 `ComponentState` per streaming message for batteries, as the `sampled_at` timestamp is attached to one `ComponentState`, we would need one for the battery state and one for the relay state.

Anyway Tiyash said he'll clarify.

shsms, 2025-02-20:
Oh I see, there's two levels of nesting.

tiyash-basu-frequenz, 2025-02-20:
At the server side, we will try our best to have each data sample as "complete" as possible, and return data for a single timestamp (with a slight catch, read below).

It means:
- All available metrics and states will be available in each data sample, provided we received them from the component hardware. E.g., battery data will contain all DC electricity samples, SoC, temperature, and all its states (overall component state and the relay state). On a side note, for all downstream client activities, component state should be all that is needed, since it also encapsulates the effect of the relay state.
- Each data sample will have metrics from a single timestamp. Again in a side note, since this results in a lot of closely spaced duplicate samples in the data stream, in future we may change this slightly (on the server implementation side) to have less duplicates. This might result in a delta of 10s of milliseconds between individual metrics. The delta might arise depending upon how the data is received from the underlying component hardware. We will still continue having all available metrics in a single message.