frequenz-floss/frequenz-api-microgrid #288: Add a tag to meter component metadata specifying if it is certified or not

tiyash-basu-frequenz, 2025-03-03:
### What's needed?

Meters placed in the same microgrid can often report different numbers for the same metric, depending upon how they are calibrated. In such cases, it is difficult for users to know which meters are more accurate than the rest. We need a mechanism to indicate if a meter produces data that is more legitimate than the others.

### Proposed solution

Some meters are certified to be used for official purposes. These meters are generally considered to be better calibrated. Perhaps adding a flag to meter metadata saying if a meter is certified or not could help the users make a more informed choice when selecting meters.

### Use cases

This would be useful while checking the grid frequency in a microgrid, since grid frequency is one of the metrics that deviate the most between meters, at the same timestamp.

Marenz, 2025-03-03:
A possible solution to this could be that we simply say, the first configured meter is considered the highest-quality meter?

tiyash-basu-frequenz, 2025-03-04:
> A possible solution to this could be that we simply say, the first configured meter is considered the highest-quality meter?

This has several challenges
- We need to define what we mean by high quality, and we need to find a definition that is consistent across all use-cases. That will be difficult. As far as this issue is concerned, we are not interested in high-resolution data, but in data that better adheres to standards. This already adds two separate axes to the definition of quality. There may be even more.
- Users might find it hard to trust a system where the quality is determined by order rather than clear indicators.
- It may create false orders. E.g., two meters could have the same "quality" (depending upon how we define it), but if we order them, then it would seem that one meter is better than the other. We want to avoid this.

IMO explicitly flagging a meter’s certification status is a more transparent indicator, as far as this issue is concerned.

That said, the point you raised about quality is a good one. I have extended the issue description with a clarification.