frequenz-floss/frequenz-api-microgrid #305: Mismatched type, `Metric` is used for filters instead of `SensorMetric`

llucax, 2025-05-13:
### What happened?

```protobuf
message ReceiveSensorDataStreamRequest {
  message SensorDataStreamFilter {
    repeated frequenz.api.common.v1.metrics.Metric metrics = 1;
  }
```

### What did you expect instead?

```protobuf
message ReceiveSensorDataStreamRequest {
  message SensorDataStreamFilter {
    repeated frequenz.api.common.v1.microgrid.sensors.SensorMetric metrics = 1;
  }
```

### Affected version(s)

v0.17 and up

### Affected part(s)

Protocol buffer definition (part:protobuf)

llucax, 2025-05-13:
[`SensorMetricSample`](https://github.com/frequenz-floss/frequenz-api-common/blob/819304a6e8f021532b84a002aba00863791c544b/proto/frequenz/api/common/v1/microgrid/sensors/sensors.proto#L252-L262) is also affected:

```protobuf
message SensorMetricSample {
  google.protobuf.Timestamp sampled_at = 1;
  frequenz.api.common.v1.metrics.Metric metric = 2;   // <-----
  frequenz.api.common.v1.metrics.MetricValueVariant value = 3;
}
```

Doing a [quick search](https://github.com/search?type=code&q=org%3Afrequenz-floss++SensorMetric), it looks like `SensorMetric` is defined but actually unused 🤷

tiyash-basu-frequenz, 2025-06-30:
This issue will be addressed implicitly only the API is migrated top common API v0.7.0 (#266)

closed 2025-07-02 by tiyash-basu-frequenz